MrChromebox/scripts #436: Cannot select 'restore stock firmware

D3dbyte, 2024-11-09:
I am trying to restore the stock os on a lenovo ideapad 3 15IGL05 and I am able to get to the part when I can choose option 2 to 'restore stock firmware' but when I type 2 and hit enter the 2 disappears and nothing happens.

I have tried this script in the terminal:

cd; curl -LO mrchromebox.tech/firmware-util.sh; sudo bash firmware-util.sh

As well as this:

cd ~; curl -LO https://mrchromebox.tech/firmware-util.sh && sudo bash firmware-util.sh

Not sure why I am having an issue.

Any help would be appreciated.

D3dbyte, 2024-11-09:
This isn't a chrome book. That's why it wouldn't work.

D3dbyte, 2024-11-09:
Customer thought this was a chromebook but I checked manufacturer site and this pc was not sold with chrome OS.

Leaving open to confirm assumption

MrChromebox, 2024-11-09:
not sure how I'm supposed to determine that given the info provided.

D3dbyte, 2024-11-09:
Sorry I used https://psref.lenovo.com/ and put in the make and model. Found that this pc was only ever sold with windows operating systems.

closed 2024-11-09 by D3dbyte