MrChromebox/scripts #437: Can't update Firmware on Copano: No chipset found

Frooastside, 2024-11-13:
I tried to update the Firmware becuase I am facing another issue where the USB-C ports don't switch to displayport alt mode automatically (i did the postinstall steps). I tried using
- ubuntu 24.04 live
- ubuntu 22.04.5 live
- fedora 41 workstation live
- fedora 41 workstation installed
- debian 12.8 live
- nixos

with every configuration neither worked i also applied the iomem=relaxed in fedora and nixos for testing, i disabled secureboot and i confirmed the hdmi adapter works with

```bash
sudo ./ectool typeccontrol 1 2 0
```
I also suspected ventoy might be an issue so i put ubuntu live on another usb stick directly.

```text
MrChromebox Firmware Utility Script starting up
 
Downloading supporting files...
 
Downloading required tools...
 
Getting device/system info...
 
Flashrom is unable to read current firmware; cannot continue:
 
flashrom MrChromebox-1.4.0-devel_2024.04.18 (git:v1.2-1418-ge5ed0c6340) on Linux 6.8.0-31-generic (x86_64)
flashrom is free software, get the source code at https://flashrom.org
 
Using clock_gettime for delay loops (clk_id: 1, resolution: 1ns).
coreboot table found at 0x76a3b000.
WARNING: No chipset found. Flash detection will most likely fail.
No EEPROM/flash device found.
Note: flashrom can never write if the flash chip isn't found automatically.
 
You may need to add 'iomem=relaxed' to your kernel parameters,
or trying running from a Live USB with a more permissive kernel (eg, Ubuntu 23.04+).
 
If you have UEFI SecureBoot enabled, you need to disable it to run the script/update your firmware.
 
MrChromebox Firmware Utility setup was unsuccessful
 * diagnosics report has been saved to /tmp/mrchromebox_diag.txt
 * go to https://forum.chrultrabook.com/ for help
 ```

MrChromebox, 2024-11-13:
> I tried to update the Firmware (current version is 2408.1)

that's the latest version

> becuase I am facing another issue where the USB-C ports don't switch to displayport alt mode automatically

that's not a firmware issue 

> WARNING: No chipset found. Flash detection will most likely fail.

never seen that before. 

COPANO is a tigerlake board, should have no issues with the version of flashrom used by the script. You could always install flashrom via the distro package manager and see if the same result using that for `sudo flashrom -p internal -r /tmp/bios.bin --ifd -i bios`

Frooastside, 2024-11-13:
> that's the latest version
ah ok i wasn't really where sure to check apart from the tool

> that's not a firmware issue
yeah i know i just wanted to give some context and thought you maybe also know something about that

> You could always install flashrom via the distro package manager
I just tried it with Fedora 41 Workstation Live, same result:
```
flashrom 1.4.0 on Linux 6.11.4-301.fc41.x86_64 (x86_64)
flashrom is free software, get the source code at https://flashrom.org
 
Cannot open file stream for /dev/mtd0
coreboot table found at 0x76a3b000.
WARNING: No chipset found. Flash detection will most likely fail.
No EEPROM/flash device found.
Note: flashrom can never write if the flash chip isn't found automatically.
```
and on ubuntu 24.04 (also same error)

but I looked for similar issues and tried it with https://mrchromebox.tech/files/util/flashrom.tar.gz from #277 and it was able to read but then segfaults??

```bash
ubuntu@ubuntu:~$ sudo ./flashrom -p internal -r /tmp/bios.bin
```

```
flashrom v0.9.9 chromium.googlesource.com/chromiumos/third_party/flashrom : 75a34ae8 : May 19 2020 02:06:04 UTC on Linux 6.8.0-31-generic (x86_64)
flashrom v0.9.9 chromium.googlesource.com/chromiumos/third_party/flashrom : 75a34ae8 : May 19 2020 02:06:04 UTC on Linux 6.8.0-31-generic (x86_64)
Calibrating delay loop... OK.
coreboot table found at 0x76a3b000.
Found chipset "Intel Tigerlake". Enabling flash write... WARNING: SPI Configuration Lockdown activated.
OK.
Reading flash... SUCCESS
Segmentation fault
```

that version doesn't seem to like the `--ifd -i bios` part though (i installed crossystem i doesnt get found and i didnt go more in that direction with that version anyways)
```bash
ubuntu@ubuntu:~$ sudo ./flashrom -p internal -r /tmp/bios.bin -i bios
```

```
flashrom v0.9.9 chromium.googlesource.com/chromiumos/third_party/flashrom : 75a34ae8 : May 19 2020 02:06:04 UTC on Linux 6.8.0-31-generic (x86_64)
flashrom v0.9.9 chromium.googlesource.com/chromiumos/third_party/flashrom : 75a34ae8 : May 19 2020 02:06:04 UTC on Linux 6.8.0-31-generic (x86_64)
Calibrating delay loop... OK.
coreboot table found at 0x76a3b000.
Found chipset "Intel Tigerlake". Enabling flash write... WARNING: SPI Configuration Lockdown activated.
OK.
sh: 1: crossystem: not found
Invalid entry specified: bios
FAILED
Segmentation fault
```

```bash
ubuntu@ubuntu:~$ sudo ./flashrom -p internal -r /tmp/bios.bin --ifd -i bios
```

```
flashrom v0.9.9 chromium.googlesource.com/chromiumos/third_party/flashrom : 75a34ae8 : May 19 2020 02:06:04 UTC on Linux 6.8.0-31-generic (x86_64)
./flashrom: unrecognized option '--ifd'
Please run "flashrom --help" for usage info.
```

MrChromebox, 2024-11-14:
> but I looked for similar issues and tried it with https://mrchromebox.tech/files/util/flashrom.tar.gz from https://github.com/MrChromebox/scripts/issues/277 and it was able to read but then segfaults (was mentioned there too)

that's a very old version of CrOS flashrom and not surprised it doesn't work on TGL

Frooastside, 2024-11-14:
> that's a very old version of CrOS flashrom and not surprised it doesn't work on TGL

Makes sense i guess, but still the only Version I found that detects the chipset, maybe I should go to their bugtracker? I am also not sure if it ever worked after the initial Installation as I said, I installed in 2024/10/10 (from chromeos) and never ran the script again until yesterday

Frooastside, 2024-11-14:
Ok after some more research i found this https://review.coreboot.org/c/flashrom/+/84987 which is made after the current flashrom release so I tried it and it actually works:

```bash
sudo builddir/flashrom -p internal -r ~/bios.bin --ifd -i bios
```

```
flashrom v1.5.0-devel (git:v1.4.0-88-ga3a8ae72) on Linux 6.11.4-301.fc41.x86_64 (x86_64)
flashrom is free software, get the source code at https://flashrom.org
 
Cannot open file stream for /dev/mtd0
coreboot table found at 0x76a3b000.
Found chipset "Intel Tiger Lake LP".
Enabling flash write... Warning: Setting BIOS Control at 0xdc from 0x8b to 0x89 failed.
New value is 0x8b.
SPI Configuration is locked down.
FREG0: Flash Descriptor region (0x00000000-0x00000fff) is read-only.
FREG1: BIOS region (0x00500000-0x01ffffff) is read-write.
FREG2: Management Engine region (0x00001000-0x004fffff) is read-only.
Not all flash regions are freely accessible by flashrom. This is most likely
due to an active ME. Please see https://flashrom.org/ME for details.
GPR0: Warning: 0x00001000-0x00198fff is read-only.
At least some flash regions are write protected. For write operations,
you should use a flash layout and include only writable regions. See
manpage for more details.
OK.
Found Winbond flash chip "W25Q256JV_M" (32768 kB, Programmer-specific) on internal.
===
This flash part has status UNTESTED for operations: WP
The test status of this chip may have been updated in the latest development
version of flashrom. If you are running the latest development version,
please email a report to flashrom@flashrom.org if any of the above operations
work correctly for you with this flash chip. Please include the flashrom log
file for all operations you tested (see the man page for details), and mention
which mainboard or programmer you tested in the subject line.
You can also try to follow the instructions here:
https://www.flashrom.org/contrib_howtos/how_to_mark_chip_tested.html
Thanks for your help!
Reading ich descriptor... done.
Using region: "bios".
Reading flash... done.
```

I moved it how it was explained in #277 to /tmp/flashrom and ran the script:

```
 ChromeOS Device Firmware Utility Script [2024-11-01] 
 (c) Mr Chromebox <mrchromebox@gmail.com> 
*********************************************************
**     Device: Asus Chromebook Flip CX5 (CX5400)
** Board Name: COPANO
**   Platform: Intel TigerLake
**    Fw Type: Full ROM / UEFI
**     Fw Ver: MrChromebox-2408.1 (09/14/2024)
**      Fw WP: Disabled
*********************************************************
** [WP] 1) Install/Update UEFI (Full ROM) Firmware 
** [WP] 2) Restore Stock Firmware 
**      C) Clear UEFI NVRAM 
*********************************************************
Select a numeric menu option or
R to reboot  P to poweroff  Q to quit 
```

Thank you for your help, by any chance do you know about where i could get help with my other problem? and also, is the dmesg output from my second comment expected or something that should not happen?

MrChromebox, 2024-11-14:
I just pushed an updated build of flashrom to my script which should resolve the issue, please reboot and try again

Frooastside, 2024-11-14:
It works correctly now

closed 2024-11-14 by Frooastside