MrChromebox/scripts #440: Firmware fails to download firmware on Asus C300M

Breakfastcrafty, 2024-11-16:
My chromebook is going to turn into a brick if it shut's down made even worse by the fact that it has a non-functioning battery when I tried to update the firmware. The error i got was: Error running cmd: /tmp/flashrom -p internal:boardmismatch=force --use-first-chip --ifd -i bios -N -w coreboot_edk2'quawks-mrchromebox_20240914.rom >/dev/null 2>&1 -o /tmp/flashrom.log

Is there any way to roll back the update or any last remaining chrome os firmware for the quawks board in the wild?

Breakfastcrafty, 2024-11-16:
Actually, the entire script isn't working for reversing using a recovery image too so I'm screwed

Breakfastcrafty, 2024-11-16:
[flashromlog.txt](https://github.com/user-attachments/files/17783553/flashromlog.txt)

MrChromebox, 2024-11-16:
> My chromebook is going to turn into a brick if it shut's down made even worse by the fact that it has a non-functioning battery when I tried to update the firmware. 

I don't know what any of this means.

why are you trying to go back to stock on an EOL device?

MrChromebox, 2024-11-17:
> I didn't intend to go back to stock. the update to the latest firmware failed so I tried to restore the stock firmware because I saw that as one of the proceedures on your site if i ever encounter an error.

you misunderstood what you read - you're supposed to flash back the firmware that was on the device at boot, which is stored by the script at launch. perhaps I should make that automatic