MrChromebox/scripts #444: Script fails to find Full ROM for Acer Chromebook 15 CB5-571 (Yuna)

leonid-bondar-d, 2024-11-19:
firmware-util.sh script is unable to find full firmware for my chromebook:

 ChromeOS Device Firmware Utility Script [2024-11-01] 
 (c) Mr Chromebox <mrchromebox@gmail.com> 
*********************************************************
**     Device: Acer Chromebook 15 (CB5-571, C910)
** Board Name: AURON
**   Platform: Intel Broadwell
**    Fw Type: Stock ChromeOS
**     Fw Ver: Google_Auron_yuna.6301.59.116 (04/30/2017)
**      Fw WP: Disabled
*********************************************************
**      1) Install/Update RW_LEGACY Firmware 
** [WP] 2) Install/Update UEFI (Full ROM) Firmware 
** [WP] 3) Set Boot Options (GBB flags) 
** [WP] 4) Set Hardware ID (HWID) 
*********************************************************
Select a numeric menu option or
R to reboot  P to poweroff  Q to quit 
2

Install/Update UEFI Full ROM Firmware

IMPORTANT: flashing the firmware has the potential to brick your device,
requiring relatively inexpensive hardware and some technical knowledge to
recover.Not all boards can be tested prior to release, and even then slight
differences in hardware can lead to unforseen failures.
If you don't have the ability to recover from a bad flash, you're taking a risk.

You have been warned.

Also, flashing Full ROM firmware will remove your ability to run ChromeOS.

Do you wish to continue? [y/N] y

NOTE: After flashing UEFI firmware, you will need to install a UEFI-compatible
OS; ChromeOS will no longer be bootable. See https://mrchromebox.tech/#faq

Press Y to continue or any other key to abort. y
The script does not currently have a firmware file for your device (AURON); cannot continue.

Press [Enter] to return to the main menu.

leonid-bondar-d, 2024-11-19:
chronos@localhost ~ $ cat /tmp/mrchromebox_diag.txt
mrchromebox firmware-util diagnostic report
Tue Nov 19 17:20:50 EET 2024

[_hwid]
AURON_YUNA_ABC-123-XYZ-456

[swWp]
disabled

[lsb-release]
CHROMEOS_ARC_ANDROID_SDK_VERSION=28
CHROMEOS_ARC_VERSION=7534053
CHROMEOS_AUSERVER=https://tools.google.com/service/update2
CHROMEOS_BOARD_APPID={FF7E5403-68B3-9E57-EE1C-A53D8AF2BAC3}
CHROMEOS_CANARY_APPID={90F229CE-83E2-4FAF-8479-E368A34938B1}
CHROMEOS_DEVSERVER=
CHROMEOS_RELEASE_APPID={FF7E5403-68B3-9E57-EE1C-A53D8AF2BAC3}
CHROMEOS_RELEASE_BOARD=auron_yuna-signed-mp-v2keys
CHROMEOS_RELEASE_BRANCH_NUMBER=97
CHROMEOS_RELEASE_BUILDER_PATH=auron_yuna-release/R91-13904.97.0
CHROMEOS_RELEASE_BUILD_NUMBER=13904
CHROMEOS_RELEASE_BUILD_TYPE=Official Build
CHROMEOS_RELEASE_CHROME_MILESTONE=91
CHROMEOS_RELEASE_DESCRIPTION=13904.97.0 (Official Build) stable-channel auron_yuna 
CHROMEOS_RELEASE_KEYSET=mp-v2
CHROMEOS_RELEASE_NAME=Chrome OS
CHROMEOS_RELEASE_PATCH_NUMBER=0
CHROMEOS_RELEASE_TRACK=stable-channel
CHROMEOS_RELEASE_VERSION=13904.97.0
DEVICETYPE=CHROMEBOOK
GOOGLE_RELEASE=13904.97.0

[boardName]
AURON

[dmidecode.device]
auron_yuna

[device]
auron

[hasUEFIoption]
true

[firmwareType]
Stock ChromeOS

[deviceCpuType.Name]
Intel Broadwell

[deviceDesc]
Acer Chromebook 15 (CB5-571, C910)

[wpEnabled]
false

[isUnsupported]
false

[fwVer]
Google_Auron_yuna.6301.59.116

[deviceCpuType.id]
BDW

[fwDate]
04/30/2017

chronos@localhost ~ $ chronos@localhost ~ $ cat /tmp/mrchromebox_diag.txt
bash: chronos@localhost: command not found
chronos@localhost ~ $ mrchromebox firmware-util diagnostic report
bash: mrchromebox: command not found
chronos@localhost ~ $ Tue Nov 19 17:20:50 EET 2024
bash: Tue: command not found
chronos@localhost ~ $ 
chronos@localhost ~ $ [_hwid]
bash: [_hwid]: command not found
chronos@localhost ~ $ AURON_YUNA_ABC-123-XYZ-456
bash: AURON_YUNA_ABC-123-XYZ-456: command not found
chronos@localhost ~ $ 
chronos@localhost ~ $ [swWp]
bash: [swWp]: command not found
chronos@localhost ~ $ disabled
bash: disabled: command not found
chronos@localhost ~ $ 
chronos@localhost ~ $ [lsb-release]
bash: [lsb-release]: command not found
chronos@localhost ~ $ CHROMEOS_ARC_ANDROID_SDK_VERSION=28
chronos@localhost ~ $ CHROMEOS_ARC_VERSION=7534053
chronos@localhost ~ $ CHROMEOS_AUSERVER=https://tools.google.com/service/update2
chronos@localhost ~ $ CHROMEOS_BOARD_APPID={FF7E5403-68B3-9E57-EE1C-A53D8AF2BAC3}
chronos@localhost ~ $ CHROMEOS_CANARY_APPID={90F229CE-83E2-4FAF-8479-E368A34938B1}
chronos@localhost ~ $ CHROMEOS_DEVSERVER=
chronos@localhost ~ $ CHROMEOS_RELEASE_APPID={FF7E5403-68B3-9E57-EE1C-A53D8AF2BAC3}
chronos@localhost ~ $ CHROMEOS_RELEASE_BOARD=auron_yuna-signed-mp-v2keys
chronos@localhost ~ $ CHROMEOS_RELEASE_BRANCH_NUMBER=97
chronos@localhost ~ $ CHROMEOS_RELEASE_BUILDER_PATH=auron_yuna-release/R91-13904.97.0
chronos@localhost ~ $ CHROMEOS_RELEASE_BUILD_NUMBER=13904
chronos@localhost ~ $ CHROMEOS_RELEASE_BUILD_TYPE=Official Build
bash: Build: command not found
chronos@localhost ~ $ CHROMEOS_RELEASE_CHROME_MILESTONE=91
chronos@localhost ~ $ CHROMEOS_RELEASE_DESCRIPTION=13904.97.0 (Official Build) stable-channel auron_yuna 
bash: syntax error near unexpected token `('
chronos@localhost ~ $ CHROMEOS_RELEASE_KEYSET=mp-v2
chronos@localhost ~ $ CHROMEOS_RELEASE_NAME=Chrome OS
bash: OS: command not found
chronos@localhost ~ $ CHROMEOS_RELEASE_PATCH_NUMBER=0
chronos@localhost ~ $ CHROMEOS_RELEASE_TRACK=stable-channel
chronos@localhost ~ $ CHROMEOS_RELEASE_VERSION=13904.97.0
chronos@localhost ~ $ DEVICETYPE=CHROMEBOOK
chronos@localhost ~ $ GOOGLE_RELEASE=13904.97.0
chronos@localhost ~ $ 
chronos@localhost ~ $ [boardName]
bash: [boardName]: command not found
chronos@localhost ~ $ AURON
bash: AURON: command not found
chronos@localhost ~ $ 
chronos@localhost ~ $ [dmidecode.device]
bash: [dmidecode.device]: command not found
chronos@localhost ~ $ auron_yuna
bash: auron_yuna: command not found
chronos@localhost ~ $ 
chronos@localhost ~ $ [device]
bash: [device]: command not found
chronos@localhost ~ $ auron
bash: auron: command not found
chronos@localhost ~ $ 
chronos@localhost ~ $ [hasUEFIoption]
bash: [hasUEFIoption]: command not found
chronos@localhost ~ $ true
chronos@localhost ~ $ 
chronos@localhost ~ $ [firmwareType]
bash: [firmwareType]: command not found
chronos@localhost ~ $ Stock ChromeOS
bash: Stock: command not found
chronos@localhost ~ $ 
chronos@localhost ~ $ [deviceCpuType.Name]

MrChromebox, 2024-11-25:
should be fixed now

closed 2024-11-25 by MrChromebox