MrChromebox/scripts #445: Flashing failure on swanky

dereference23, 2024-11-24:
```
 ChromeOS Device Firmware Utility Script [2024-11-01] 
 (c) Mr Chromebox <mrchromebox@gmail.com> 
*********************************************************
**     Device: Toshiba Chromebook 2 (2014) CB30/CB35
** Board Name: SWANKY
**   Platform: Intel BayTrail

 ChromeOS Device Firmware Utility Script [2024-11-01] 
 (c) Mr Chromebox <mrchromebox@gmail.com> 
*********************************************************
**     Device: Toshiba Chromebook 2 (2014) CB30/CB35
** Board Name: SWANKY
**   Platform: Intel BayTrail
**    Fw Type: Full ROM / UEFI
**     Fw Ver: MrChromebox-4.20.0 (05/15/2023)
**             Update Available (09/14/2024)
**      Fw WP: Disabled
*********************************************************
```

[flashrom.log](https://github.com/user-attachments/files/17894599/flashrom.log)

MrChromebox, 2024-11-24:
seems the new flashrom binary I'm using isn't playing nicely with Baytrail/Braswell devices, I'm looking into it.

For now, if you haven't exited and re-run the script, you can restore the previous firmware by running:

`sudo /tmp/flashrom -p internal -w /tmp/bios.bin --ifd -i bios -N`

after which you can reboot

dereference23, 2024-11-25:
I left my laptop at sleep mode (I didn't turn it off) and now it's bricked. I will have to restore it in some other way...

MrChromebox, 2024-11-25:
https://docs.mrchromebox.tech/docs/support/unbricking/unbrick-ch341a.html

MrChromebox, 2024-12-21:
flashrom issue has been fixed

closed 2024-12-21 by MrChromebox