MrChromebox/scripts #451: CX12 won't boot after script

The-Captain56, 2024-12-25:
I own a Chromebox CX12 & followed your script on how to convert it to a normal computer. The script ran fine. At the menu there was a boot option, however however I exited the menu & the boot menu option the next screen had no boot option. Didn't know how to get back to the prior screen, Alt+Ctrl + F2 and other commands didn't work. I shut it down thinking it would reboot to the menu screen but w/ no OS I just get a blank monitor. If I power it on manually I get a msg that states no analog input. Tried using the recovery USB and a USB containing bootable Cinnamon same result. The CX12 pwrs up but the screen is blank.
               How do I boot this w/ no OS? Can't do anything w/ a blank monitor. 
I've tried CTRL+Alt+Refresh; 
Refresh + Power for 10 sec; 
Esc +Refresh +Pwr.

Thank you,
 The Captain

MrChromebox, 2024-12-26:
>  If I power it on manually I get a msg that states no analog input. 

the CXI2 does not have analog outputs, your monitor is likely set to the wrong input.

> I've tried CTRL+Alt+Refresh;
> Refresh + Power for 10 sec;
> Esc +Refresh +Pwr.

these do nothing on a Chromebox

The-Captain56, 2024-12-26:
Are there any keystokes that will allow it to get back to the OS  boot optionoption. This is the same monitor that it always used w/ the same input. It is currently working for another laptop. I also tried the CX12 on a different monitor - same story

David A. Kenney

MrChromebox, 2024-12-26:
you press ESC when the boot splash shows and the text 'Press ESC for Boot Options" shows at the bottom of the screen.

If the Chromebox worked after flashing initially but isn't now, I wouldn't suspect an issue with the box itself/firmware. It's something with the display connection/handshaking

The-Captain56, 2024-12-27:
Nothing shows On screen as I initially stated. I need to get past the connection issue therefore back to the boot keystroke question. Is there a combination of keystrokes to get to the OS boot screen?

David A. Kenney

MrChromebox, 2024-12-27:
if nothing shows on screen, then no key sequence is going to help - you need to unbrick/re-flash the box per my link above