MrChromebox/scripts #453: Pixelbook (EVE) won't boot ChromeOS after restore to stock firmware

simonepsp, 2024-12-30:
Hello everyone,
as title says I have been unable to boot ChromeOS after restoring to stock firmware on my Pixelbook 2017 (eve). You can find a screenshot of the error below ("no bootable kernel found on disk"). There seems to be a TPM issue as well?

It comes without saying that I've already attempted (multiple times and using multiple usb drives) to restore the device using the Chromebook recovery utility. All attempts were **successful** and upon removing the recovery drive the Pixelbook successfully rebooted but it was still unable to boot ChromeOS. 

Interestingly enough, I've also tried to revert to MrChromebox UEFI ROM and everything worked just fine (including installing a Linux distro on the internal SSD and booting from it!). So I guess there's no HW issue... 

What I tried:
-> Restoring my stock BIOS backup, saved with the MrChromebox utility
-> Extracting the EVE / Pixelbook board firmware included in the recovery image and flashing it manually through flashrom. 
-> Resetting the TPM by running "tpm2_clear" on ArchLinux (while on UEFI rom bios)
-> I could confirm that all the boot entries (local image A, local image B, verified image A, verified image B) were correctly created. Of course I can't boot from them, but I could see those GRUB entries while being on UEFI rom.

Should it be helpful, I have at my disposal a Suzy-Q cable as well (that I've used for the flash process).
Thank you

[image]

MrChromebox, 2024-12-30:
> I could confirm that all the boot entries (local image A, local image B, verified image A, verified image B) were correctly created. Of course I can't boot from them, but I could see those GRUB entries while being on UEFI rom.

these are not used by the stock firmware - the stock firmware boots the ChromeOS kernel directly, doesn't use GRUB in any way

>  All attempts were successful and upon removing the recovery drive the Pixelbook successfully rebooted but it was still unable to boot ChromeOS.

The error you're showing above means the firmware can't boot the kernel - either it's not found, or failing verification for some reason.

simonepsp, 2024-12-30:
> these are not used by the stock firmware - the stock firmware boots the ChromeOS kernel directly, doesn't use GRUB in any way
>
>
>
>
>
>
> The error you're showing above means the firmware can't boot the [...]

Thank you for looking into it.
Do you have any clue on why this is happening? 

I guess there's no reason why the kernel wouldn't be there...right?