MrChromebox/scripts #454: Won’t boot without internal display connected on Samsung Chromebook 2 11 - XE500C12 Chromebook Winky after firmware update

henny-0, 2024-12-30:
Just updated to the latest MrChromebox firmware (MrChromebox-2408.1) for my Chromebook and and realized through trial that the device won’t boot without the internal display connected. Prior to this I was on firmware version 4.20.1 and could boot with just the HDMI connected to a monitor (not sure about headless booting). This would also show the Bios screen.

Now it just powers up (shown by the led indicator) if the internal display isn’t connect, but there is no indication of anything else happening. Pressing the power button once even after 5 minutes turns off the led. HDMI shows no indication of receiving a signal through all this period.

If I power down and attach the internal display, everything works. HDMI output operates like it did with the screen attached before, only after the bios screen.

MrChromebox, 2024-12-30:
hmm, I'm not sure why that would be, I'd have to look at the display init type used for Winky. Nothing has changed on Baytrail devices for some time so I'm surprised to see this regression

henny-0, 2025-01-03:
Thank you for your response. To be honest I thought it was dead. The internal display is left unconnected because its damaged.