MrChromebox/scripts #457: Error: failed to extract firmware image for phaser360 using this recovery USB

TGGBots, 2025-01-07:
I saw a similar reddit thread about this issue before with a lenovo 300e I think it may have reappeared this is what happens after it says it's extracting.

```
cat: /tmp/tmp.GGGaZFrQt4/VERSION: No such file or directory
cp: cannot stat '/tmp/tmp.GGGaZFrQt4/bios.bin': No such file or directory
Error: failed to extract firmware image for phaser360 using this recovery USB
```

running Manjaro

MrChromebox, 2025-01-07:
I just pushed an update to the script which should hopefully resolve this - I tested it "offline" and it worked for phaser360. Give it 5-10 mins for GH to sync, then try re-running the script and restore from the recovery USB

TGGBots, 2025-01-07:
I'm restoring via a chrome os recovery usb and it still doesn't seem to want to work should I install a different operating system? There is also an update for the fw if that's of any use other than that I can't see any other reason.

MrChromebox, 2025-01-07:
> it still doesn't seem to want to work

I can't help with such vagaries. Exact error output from the script please

TGGBots, 2025-01-07:
It was the same error as last time.
`cat: /tmp/tmp.GGGaZFrQt4/VERSION: No such file or directory
cp: cannot stat '/tmp/tmp.GGGaZFrQt4/bios.bin': No such file or directory
Error: failed to extract firmware image for phaser360 using this recovery USB`

MrChromebox, 2025-01-07:
I don't understand why you did the `ls -lart` on a different directory than what the script gave in the error after re-running it, that's not particularly helpful.

I've pushed an update which should provide more detail in the error output, including listing the directory contents, which hopefully should help me figure out what's going wrong

TGGBots, 2025-01-08:
I should've provided context the `ls -lart` is from a previous attempt:

[image]
as for the error you were right it is the recovery drive. 

[image]
Which is weird because I used the bin file from cros.tech like in the tutorial 

and the same error happened when I was using the chromebook recovery utility 
(Note this was from before you added the extra debugging features it was just that I got the file not found error that was the same as when I used the .bin file)

although I downloaded these with brave which I've had trouble with it corrupting my downloads (although that was on windows so idk)

sorry for the late reply btw.

MrChromebox, 2025-01-08:
I will have to dig into this here and see why there are no files in the dir, and why it works with the standalone coreboot script

MrChromebox, 2025-01-08:
ok, I just tested here using an octopus recovery image written to USB, and everything works properly when executing the script function manually. The only thing I can think of is that correct the USB device isn't being selected from the prompt, or somehow the numeric selection isn't translating to the correct device

MrChromebox, 2025-01-08:
do me a favor and attach the file `/tmp/mrchromebox_diag.txt` after running the script

TGGBots, 2025-01-08:
I got this output:
[image]
I tried looking through the partitions after and one cannot be opened so I'm going to try to re-download the image maybe that's it

MrChromebox, 2025-01-08:
the only partition we care about is p3, and I'm guessing it's fine. The script is using debugfs to grab  `/usr/sbin/chromeos-firmwareupdate` from `/dev/sdc3`, and that appears to be what's failing. No idea why though. You could tmp mount it and check that the file exists

TGGBots, 2025-01-09:
ok so I've tried mounting it in dolphin and in the shell and I get this:
[image]
is it corrupted or do I need to install something to make it read-able I've redownloaded this image twice and used the chromebook repair thing online.

although I guess this means that it isn't the firmware.

 What should I do now consider?

MrChromebox, 2025-01-09:
I think the issue is that p3 is a Cros partition not a regular ext3/4 one, and that's why the script uses debugfs to extract the updater vs a normal mount operation. I got the same error here mounting under pop. try using `sudo mount -o,ro /dev/sdc3 ...` -- that worked for me

TGGBots, 2025-01-09:
No dice same error, could you add a verbose mode that would probably help because I have no idea when something may have helped.

MrChromebox, 2025-01-09:
it would take a lot of work to implement that over the entire script. You can you live boot Ubuntu 2404 and try from there?

TGGBots, 2025-01-10:
I installed Ubuntu and then found the problem.

Out of desperation I updated the mrchromebox firmware on the device which made the rest work

Considering what errors we were getting I wasn't exactly thinking that was a possibility I'm sorry for the trouble thanks though

closed 2025-01-10 by TGGBots

MrChromebox, 2025-01-10:
> I installed Ubuntu and then found the problem.

which was? there's no way the firmware version of the device affected the extraction of the firmware from the recovery USB

TGGBots, 2025-01-10:
I know it shouldn't matter if it's updated or not that's why I didn't try it sooner but apparently it does, idk why.