MrChromebox/scripts #461: Acer Spin 312 (CRAASKOV) not recognized

onedayfishsale, 2025-01-26:
Not sure whether to post this here or in the scripts repo, but running the 2025-01-05 firmware utility script on my Acer Chromebook Spin 312 detects it as a Spin 512 and says "MrChrombox Firmware Utility didn't recognize your device". Diagnostics report is attached.

[mrchromebox_diag.txt](https://github.com/user-attachments/files/18547759/mrchromebox_diag.txt)

MrChromebox, 2025-01-26:
technically it's both - the script doesn't support the device because there's no firmware

this device isn't in any of the recovery lookups yet, so in order to add support I'd need your stock firmware. When you run the script, the firmware gets saved to /tmp/bios.bin. copy that file to USB (probably need to manually mount it; you can scp it somewhere too if easier), zip it, and attach here.

onedayfishsale, 2025-01-26:
Thanks for moving the issue. Zipped bios.bin attached.

[bios.zip](https://github.com/user-attachments/files/18548623/bios.zip)

MrChromebox, 2025-01-26:
ugh just realized this is only the bios region - can you do a manual read for me?

`sudo flashrom -p internal -r craaskov_stock_internal.rom`

onedayfishsale, 2025-01-26:
Yep.

[craaskov_stock_internal.zip](https://github.com/user-attachments/files/18548729/craaskov_stock_internal.zip)

MrChromebox, 2025-01-26:
just pushed a script update to add support, as well as a Full ROM image. Since it's a new board and untested, I would only recommend if you have a SuzyQ / are able to unbrick if something goes awry

onedayfishsale, 2025-01-26:
Thanks, I do have a SuzyQ and will test it when I get a chance.

onedayfishsale, 2025-01-27:
Flashed the UEFI ROM and it appeared to succeed but I just get a black screen at power on. Will read up on debugging and see what I can see.

MrChromebox, 2025-01-28:
> Flashed the UEFI ROM and it appeared to succeed but I just get a black screen at power on. Will read up on debugging and see what I can see.

might need to have you flash a debug ROM with serial output so we can see what's going on, if simply reflashing via suzyQ doesn't resolve it

onedayfishsale, 2025-01-28:
Flashed the stock ROM back successfully, so not bricked. Will try the UEFI ROM via the normal script again here shortly.

MrChromebox, 2025-01-28:
that's unlikely to produce a different result, I'd recommend flashing a debug build instead

onedayfishsale, 2025-01-28:
Sorry, misunderstood what you said before. If you have a ROM for me to try I'll give it a shot.

MrChromebox, 2025-01-29:
here's a debug/serial-enabled build which can be flashed manually:

[coreboot_edk2-craaskov-mrchromebox_20250128.zip](https://github.com/user-attachments/files/18581295/coreboot_edk2-craaskov-mrchromebox_20250128.zip)

`sudo flashrom -p internal -w <filename> --ifd -i bios -N`

use the SuzyQ connected to another device to record the console log output on ttyUSB1

onedayfishsale, 2025-01-29:
Serial log attached. Seems to hang at `[DEBUG]  Jumping to boot code at 0x008016da(0x769fa000)`.

[craaskov_serial_log.txt](https://github.com/user-attachments/files/18587314/craaskov_serial_log.txt)

MrChromebox, 2025-01-29:
that's not a hang, that's the end of the log.

The initial Craaskov image was missing the VBT file necessary to init the display/IGD, everything else is fine.

If you restore to stock and reflash from the script, everything should be good to go now (fingers crossed)

onedayfishsale, 2025-01-30:
Worked! Booted fully, testing in Fedora 41 for now. Thanks for adding support and for your help with flashing.

onedayfishsale, 2025-01-30:
On the off chance it might be a coreboot thing, I'll just mention that currently audio isn't working. Still chasing error messages though, will update if I find something.

MrChromebox, 2025-01-30:
> On the off chance it might be a coreboot thing, I'll just mention that currently audio isn't working. Still chasing error messages though, will update if I find something.

nothing to do with the firmware, everything to do with the difference in audio subsystems bteween ChromeOS and mainline linux -- see the chrultrabook docs for a post-install script which may help

closed 2025-02-03 by onedayfishsale

Fox8830, 2025-11-29:
Very sorry to talk in this thread after it has been closed, but does this bring resolved mean CRAASKOV boards are supported?
I don't see them in the list of supported devices, but from reading this it seems to be working, so i thought id ought to check.
Apologies if ive missed something that would answer my question.

onedayfishsale, 2025-11-29:
> Very sorry to talk in this thread after it has been closed, but does this bring resolved mean CRAASKOV boards are supported?

It's working great for me. Sample size of one and everything, but I just upgraded to the latest firmware yesterday and it continues to work fine.

Fox8830, 2025-11-29:
Was looking on docs.chrultrabook.com, it seems to be absent ther. Sorry, and thanks for both the software and the reply, much appreciated!!

MrChromebox, 2025-11-29:
> Was looking on docs.chrultrabook.com, it seems to be absent there

they need to sync their docs with mine more often then :)