MrChromebox/scripts #462: Failed to extract shellball from chromeos-firmwareupdate

Nima1279, 2025-01-28:
Dear MrChormebox
I have been tried to restore stock the firmware on HP Chromebook x360 14 but I can't !!!
when I try to restore the firmware it leads to Error: failed to extract firmware for SONA from this ChromeOS recovery USB 
also I tried many ways to fix it such as getting the recovery firmware on your Github and changing the usb and........but they didn't work.
thanks for suggestions

Etm571, 2025-02-18:
Hello,

I have the same issue. Have tried:

Creating a recovery USB through Google's official browser tools

Flashed the .bin file downloaded from cros.tech(mentioned in the documentation) to a USB drive with both Rufus and Balena Etcher.

For the live linux environment I'm using the Arch Linux installer ISO flashed to another USB drive plugged in to the system. I'm not sure if Arch is supported and/or it could cause any problems...

[image]

Chromebook: Toshiba Chromebook 2 CB30 2014(SWANKY)

MrChromebox, 2025-02-18:
@Etm571 why are you trying to go back to EOL ChromeOS on Swanky?

Etm571, 2025-02-19:
@MrChromebox 

I need to reinstall the custom firmware because I messed up the boot sequence. I can no longer boot from SD cards. I've tried to add a boot option and boot from file, but those don't appear to be working for me..

MrChromebox, 2025-02-19:
@Etm571 ok, but that doesn't answer my question at all. Clearing the boot sequence is as simple as using the Clear NVRAM option on the script. Reinstalling the firmware (which isn't needed and doesn't clear the boot sequence) doesn't require going back to stock. You never need to go back to stock.

Nima1279, 2025-02-19:
@Etm571 
Thanks for your suggestion  
You mean I have to format the usb with Rufus then download the chrome os on Google's official tools after that Reinstalling the firmware?
can you explain it more and how can you fix this same problem?

Etm571, 2025-02-19:
@Nima1279 

You need to download the correct .bin file for your Chromebook model. Then you have to flash the .bin file with Rufus according to the docs. But that didn't solve it for me, but maybe it solves it for you:)

Etm571, 2025-02-19:
Ok thanks! Will definitely try that> @Etm571 ok, but that doesn't answer my question at all. Clearing the boot sequence is as simple as using the Clear NVRAM option on the script. Reinstalling the firmware (which isn't needed and doesn't clear the boot sequence) doesn't require going back to stock. You never need to go back to stock.

Etm571, 2025-02-19:
@MrChromebox 

Ok thanks! Will definitely give that a try!

MrChromebox, 2025-02-19:
@Etm571 please don't flash the stock firmware, just clear the NVRAM and reboot.

Nima1279, 2025-02-19:
@MrChromebox 
How can we clear the NVRAM?
If I clear the  NVRAM is it going back to the original chrome os? 
because when I search about clear NVRAM it's like powerwash and if you install another firmware you can't switch to the original firmware by clearing the 
  NVRAM!!

MrChromebox, 2025-02-19:
@Nima1279 it's literally in the script documentation on my site:

> If running UEFI Full ROM firmware, there will be an additional option to clear the NVRAM. This will delete all bootorder entries stored in NVRAM, and they will be created again on the next boot (or next time grub is updated).

it has nothing at all to do with your issue. I don't know why it's failing to extract the firmware for SONA, it works fine here when running the function locally using a NAMI recovery image. 

[coreboot-Google_Nami.10775.123.0.zip](https://github.com/user-attachments/files/18872530/coreboot-Google_Nami.10775.123.0.zip)

dacampsss, 2025-02-19:
I'm having this same issue while trying to go to stock, but with a kasumi (Stoney Ridge) device. I'm sure that I downloaded the right firmware from cros.tech, as adviced (and flashed it using dd). I'm running Debian testing (trixie, 13). Output is below:

`
 ChromeOS Device Firmware Utility Script [2025-01-05] 
 (c) Mr Chromebox <mrchromebox@gmail.com> 
*********************************************************
**     Device: Acer Chromebook 311
** Board Name: KASUMI
**   Platform: AMD StoneyRidge
**    Fw Type: Full ROM / UEFI
**     Fw Ver: MrChromebox-2408.1 (09/14/2024)
**      Fw WP: Disabled
*********************************************************
** [WP] 1) Install/Update UEFI (Full ROM) Firmware 
** [WP] 2) Restore Stock Firmware 
**      C) Clear UEFI NVRAM 
*********************************************************
Select a numeric menu option or
R to reboot  P to poweroff  Q to quit 
2

Restore Stock Firmware

Standard disclaimer: flashing the firmware has the potential to
brick your device, requiring relatively inexpensive hardware and some
technical knowledge to recover.  You have been warned.

Do you wish to continue? [y/N] y

Please select an option below for restoring the stock firmware:

1) Restore using a firmware backup on USB
2) Restore using a ChromeOS Recovery USB
Q) Quit and return to main menu

? 2

Connect a USB which contains a ChromeOS Recovery Image
and press [Enter] to continue. 

Devices available:

1) USB_DISK_2.0 (14.46 GB)

Enter the number which corresponds your ChromeOS Recovery USB: 1

Using USB device: /dev/sda

Extracting firmware from recovery USB

Failed to extract shellball from  'chromeos-firmwareupdate'

Error: failed to extract firmware for KASUMI from this ChromeOS recovery USB

Press [Enter] to return to the main menu.
`

lsblk output:
`
NAME         MAJ:MIN RM  SIZE RO TYPE MOUNTPOINTS
sda            8:0    1 14,5G  0 disk 
├─sda1         8:1    1  3,3M  0 part 
├─sda2         8:2    1  512M  0 part 
├─sda3         8:3    1    4G  0 part 
├─sda4         8:4    1   16M  0 part 
├─sda5         8:5    1    2M  0 part 
├─sda6         8:6    1  512B  0 part 
├─sda7         8:7    1  512B  0 part 
├─sda8         8:8    1   16M  0 part 
├─sda9         8:9    1  512B  0 part 
├─sda10        8:10   1  512B  0 part 
├─sda11        8:11   1    8M  0 part 
└─sda12        8:12   1   32M  0 part 
mmcblk1      179:0    0 29,7G  0 disk 
└─mmcblk1p1  179:1    0 29,7G  0 part /media/daxxx/3566-3031
mmcblk0      179:8    0 29,1G  0 disk 
├─mmcblk0p1  179:9    0  976M  0 part /boot/efi
└─mmcblk0p2  179:10   0 28,1G  0 part /
mmcblk0boot0 179:16   0    4M  1 disk 
mmcblk0boot1 179:24   0    4M  1 disk 
zram0        254:0    0  1,9G  0 disk [SWAP]
`

uname -a output:
`
Linux debianbook 6.6.68-chrultrabook-stoney #1 SMP PREEMPT_DYNAMIC Fri Dec 27 23:40:47 UTC 2024 x86_64 GNU/Linux
`

I can give more logs and outputs, if needed to solve this issue.
Thanks in advance!

dacampsss, 2025-02-19:
Ps.: I tried both the latest 132x branch/version as well as the oldest available (in cros website) 123x version.

MrChromebox, 2025-02-19:
@dacampsss I don't have a good way to debug this, just use the coreboot crosfirmware.sh script to do the same thing:

https://github.com/coreboot/coreboot/tree/main/util/chromeos

then throw the firmware on USB and restore that way.

dacampsss, 2025-02-20:
> [@dacampsss](https://github.com/dacampsss) I don't have a good way to debug this, just use the coreboot crosfirmware.sh script to do the same thing:
> 
> https://github.com/coreboot/coreboot/tree/main/util/chromeos
> 
> then throw the firmware on USB and restore that way.

Thanks so much for your  help! I was able to restore Google firmware using this. Unfortunately, I'm facing an error now that unables me to flash the firmware, as it seems. After plugging in my USB drive with a firmware image, I get an error "The device owner requires this device to run with ChromeOS verification turned on", and the Ctrl+O output tells me that verification is off. However, I can't enable or disable the developer mode, cause pressing Ctrl+D in the recovery screen gave me no output or response. I've already tried both a dd'ed cros.tech image, and also a USB stick created using Google's linux_recovery script.

I think this may be an issue related to the battery unlock method, so I will open up my Chromebook again, remove battery and try the reset. If this also not works, I'll try to create a Windows (or openFyde?) VM to use the "official" recovery USB creator by Google.

Nima1279, 2025-02-21:
@MrChromebox 
You know I've wanted to reinstall the firmware becouse after installing the new firmware the speaker and backlight of the Chromebook stoped working...
Do you have an idea to solve this issue?

MrChromebox, 2025-02-21:
> [@MrChromebox](https://github.com/MrChromebox) You know I've wanted to reinstall the firmware becouse after installing the new firmware the speaker and backlight of the Chromebook stoped working... Do you have an idea to solve this issue?

same recommendation as above

dacampsss, 2025-02-21:
> > [@dacampsss](https://github.com/dacampsss) I don't have a good way to debug this, just use the coreboot crosfirmware.sh script to do the same thing:
> > https://github.com/coreboot/coreboot/tree/main/util/chromeos
> > then throw the firmware on USB and restore that way.
> 
> Thanks so much for your help! I was able to restore Google firmware using this. Unfortunately, I'm facing an error now that unables me to flash the firmware, as it seems. After plugging in my USB drive with a firmware image, I get an error "The device owner requires this device to run with ChromeOS verification turned on. You are trying to install an unofficial ChromeOS recovery image which does not pass verification", and the Ctrl+O output tells me that verification is off. However, I can't enable or disable the developer mode, cause pressing Ctrl+D in the recovery screen gave me no output or response. I've already tried both a dd'ed cros.tech image, and also a USB stick created using Google's linux_recovery script.
> 
> I think this may be an issue related to the battery unlock method, so I will open up my Chromebook again, remove battery and try the reset. If this also not works, I'll try to create a Windows (or openFyde?) VM to use the "official" recovery USB creator by Google.
> 
> Ps.: removing the battery itself didn't work. I'll try doing things with my USB stick now (and keep the battery removed).

The second solution worked well (create an USB stick using the official extension, within a Windows kvm/qemu VM, and not dd or the Linux script).

Nima1279, 2025-02-23:
@dacampsss 
Can you explain more about your second solution and what do you mean about Windows kvm/qemu VM?

dacampsss, 2025-02-23:
> [@dacampsss](https://github.com/dacampsss) Can you explain more about your second solution and what do you mean about Windows kvm/qemu VM?

I'll try to write briefly on what I did, step by step.

1. In my Fedora installation, I ran the coreboot script that mrchromebox pointed above to automatically download and extract the firmware file. The script will ask you a name that is your device's codename (in my case, Kasumi360), not the board name (Grunt, in my case) or the device model (Spin R721T in my case);
2. Then, I moved the file to the root of my USB stick, ran the firmware utility script by mrchromebox, and selected the option to restore stock using a firmware backup (despite this being not a backup exactly, but a firmware dropped by the coreboot script);
3. After that, I faced the issue I described, that any firmware USB generated using dd command (or Rufus, it could be) or even the firmware script for Linux made available by Google triggered an error in the recovery. What I did to solve it was, basically, create the USB stick using the official Google extension for creating recovery ChromeOS sticks, available only for Windows, Mac and ChromeOS, not Linux. Instead of installing an entire Windows installation in my real hardware ("bare metal"), I've made it through a KVM/qemu Virtual Machine running Windows 10. KVM/qemu is a Linux specific solution for running virtual machines, that have advanced features and more deep access to the host's hardware as it runs in kernel space, not user space (as well as VMWare, that is more friendly for end users, but closed source).

nirob13-12, 2025-07-06:
In my Chromebook I installed Windows now I did what I need to do to return to my Chrome OS and I did it but some problems are mentioned below ==. == failed to extract shellball from 'chromeos-firmwareupdate' error: failed to extract firmware for treeya from this chromeos recovery usb

Chromebook Model = Lenovo 100e 2nd gen and Treeya 

[image] 

[image]