MrChromebox/scripts #464: Bricked hp x360 11 G3 EE

evloebenstein, 2025-02-22:
I first installed legacy Firmware directly to the eeprom with a ch341 and 1.8v module board without de soldering the chip but after seeing that the firmware didn't flash correctly i decided to de solder the eeprom and flash it directly to it .
This managed to flash the rom but when I soldered it back and turned on the Chromebook it showed a black screen. 
First I thought it was my soldering so I decided to check it and also the eeprom to see if it didn't burn or touched some resistance nearby.
The chip was still working, it read fine and I also managed to rollback the rom But it still won't show image.

evloebenstein, 2025-02-22:
The Chromebook after flashing the rom it had the board lights functioning without any flicker so it isn't a board problem

MrChromebox, 2025-02-22:
> I first installed legacy Firmware directly to the eeprom with a ch341 

this doesn't tell me what you actually did. tell me what exactly file you flashed, and what flashrom command line