MrChromebox/scripts #465: Firmware update not working on Linux on SASUKE without `--fmap` flag

Frollamma, 2025-03-21:
I installed the Tianocore firmware for SASUKE using the `firmware-utils.sh` script on ChromeOS, it worked without any issues. Today I tried to update the firware on Void Linux and I found two issues:
1. If I selected "1", it would just go back to `stock_menu`, causing a loop. So I edited `firmware.sh` and forced it to run the `flash_rwlegacy` function
2. `flash_rwlegacy` didn't work and I got an error like `region requested with -i RW_LEGACY but no layout data is available`. I got that the `flashrom` command couldn't find the `RW_LEGACY` region and I fixed it by adding `--fmap` to it, so I substituted this line
```
if ! ${flashromcmd} -w -i RW_LEGACY:${rwlegacy_file} -o /tmp/flashrom.log > /dev/null 2>&1; then
```
with this line
```
if ! ${flashromcmd} --fmap -w -i RW_LEGACY:${rwlegacy_file} -o /tmp/flashrom.log > /dev/null 2>&1; then
```
I got to this conclusion because the problem was a missing layout file and, by reading the [flashrom manual](https://flashrom.org/classic_cli_manpage.html), it turns out that there are different flags that can be used to read layout files from chip, I tried `--ifd` and `--fmap`. Before writing to the chip I run `flashrom -x --fmap`, that extracted the regions contents in the current directory and I got a file named `RW_LEGACY` (probably `flashrom -r --fmap` should be enough to list that region). At this point I was sure that the right region was there and I wrote to the chip using the edited script.

MrChromebox, 2025-03-21:
> Today I tried to update the firware on Void Linux and I found two issues:

ok, why? RWL for JSL hasn't been updated since it was released in 2021

> I got to this conclusion because the problem was a missing layout file

I changed the flashrom binary from Chromium to upstream, and --fmap is no longer implied, has to be explicit. Looks like I missed added it here

closed 2025-03-21 by MrChromebox

Frollamma, 2025-03-21:
> ok, why?
Cause I didn't know when the last update was done

commit 8f9eb5be (2025-03-21): firmware.sh: Use --fmap for RWL flashing under Linux

Needed by upstream flashrom. Fixes #465