MrChromebox/scripts #469: Couldn’t resolve host

samson2854, 2025-03-27:
PARROT SICKLEBILL A-D 1328
I have done the jumpers
My hope is to flash the firmware and turn this older chrome book into a regular PC with a Lubuntu install.

[image]

What am I getting wrong? I wasn’t sure about the space after -LO or  -LOk, so I tried it many ways.
Thank you, Mr. Chromebox I’m sure this will be fantastic after I get it working.

samson2854, 2025-03-27:
After looking much closer, I see my continued mistake of not putting the "e" at the end of "chrome". So now I'm getting a different error. Regarding "unable to get local issue certificate"

MrChromebox, 2025-03-27:
I'm not sure how the documentation on my site isn't clear on the command line to use:
https://docs.mrchromebox.tech/docs/fwscript.html

it even tells you how to deal with SSL certificate errors on older devices. Typos I can't help with though.

samson2854, 2025-03-27:
Mr.Chromebox thank you for your guidance, sorry about the silly typing errors. I saw in your final comment in your utility that the first boot could take a long time. It has been more than an hour, I am using a Ventoy to boot a Lubuntu.iso. Should I make a regular USB Lubuntu key?

MrChromebox, 2025-03-27:
a long time = up to 90 seconds. PARROT hasn't been tested in some time, possible it's no longer working. But I would disconnect the battery for 30s and then try booting again before trying to unbrick it

samson2854, 2025-03-27:
Made a regular USB boot key. Everything's fine. Thank you very much. This is just awesome.

closed 2025-03-27 by MrChromebox