MrChromebox/scripts #471: Error downloading one or more required files; cannot continue

fortune13333, 2025-04-04:
after typing cd; curl -LO mrchromebox.tech/firmware-util.sh && sudo bash firmware-util.sh

warn me
MrChromebox Firmware Utility Script starting up

Downloading supporting files...
Error downloading one or more required files; cannot continue

fortune13333, 2025-04-04:
[image]

fortune13333, 2025-04-04:
my device is lenovo thinkpad 13 chromebook, i have remove wp screw

MrChromebox, 2025-04-04:
the files are being downloaded from github, are you able to access github.com?

fortune13333, 2025-04-04:
i can

> the files are being downloaded from github, are you able to access github.com?

fortune13333, 2025-04-04:
i even download firmware-util,sh and the other 3 .sh file ,an put them into a same filefolder ,"sudo bash  the firmware-util.sh" still dont work.
but when using "bash firmware-util.sh" it will work ，it will download the other 3 files But at the end, it will prompt, please use root privileges, such as sudobash......，

MrChromebox, 2025-04-04:
> i even download firmware-util,sh and the other 3 .sh file ,an put them into a same filefolder ,"sudo bash the firmware-util.sh" still dont work.

well, that's not how the script works, so not sure why you'd expect it to

make sure you are on the latest ChromeOS version for your board, perform a USB recovery, and try again