MrChromebox/scripts #472: TARLO - missing firmware file

Rougnt, 2025-04-06:
I'm trying to update my firmware by ChromeOS Device Firmware Utility Script.  Write protect is disabled. I downloaded and ran the firmware script from within a Fedora.
Error message is 
```
The script does not currently have a firmware file for your device (TARLO); cannot continue.
```

MrChromebox, 2025-04-06:
I'll have to add it as part of the upcoming 25.03 release

Rougnt, 2025-04-07:
> I'll have to add it as part of the upcoming 25.03 release

Thank you for your response and for planning to add support for my device in the upcoming 25.03 release.

Rougnt, 2025-04-08:
> so what board name does the script report when run from Fedora? If it's not running my firmware, then whose is it running?

The script report the board name is "TARLO".
I think the firmware I'm running is still yours. Because they have the same menu items in the BIOS.

<details>
<summary>Script output</summary>

```
**     Device: Lenovo 5i Chromebook 16\"
** Board Name: TARLO
**   Platform: Intel AlderLake/RaptorLake-U/P
**    Fw Type: Full ROM / UEFI
**     Fw Ver: 4.21-199-g1703646268yh (11/01/2023)
**      Fw WP: Disabled
```

</details>

MrChromebox, 2025-04-09:
the config for this firmware is bizarre. commit hash doesn't match anything in my repo. defines Tarlo as a separate board, but doesn't exist as such in coreboot (and never has, AFAICT). Uses binaries from Taeko.

Rougnt, 2025-04-10:
OK, Thank you for your help. Will this be addressed in 25.03 release? Or should I manually flash Taeko's UEFI firmware?

MrChromebox, 2025-04-10:
I'll add a Tarlo-specific build for 2503

Rougnt, 2025-04-11:
Thank you again for your help.

closed 2025-04-11 by Rougnt