MrChromebox/scripts #473: DOCHI - missing firmware

logicone96, 2025-04-28:
I'm attempting to install the full ROM firmware on my Acer Chromebook Plus Spin 514 (CP514-4HN) 
Write protect is disabled. I downloaded and ran the firmware script from within a ChromeOS Guest login.

Error message is: "The script does not currently have a firmware file for your device (DOCHI); cannot continue".

Spec:
Acer Chromebook Plus Spin 514
CPE-594-1N-TCO-52GR
SNID: 50310924376
Board: DA0ZCQMBAD0 Rev:D , model: ZCQ

Data from script:
Device: Acer Chromebook Plus Spin 514 (CP514-4HN)
Board Name: DOCHI
Platform: Intel AlderLake/RaptorLake-U/P
Fw Type: Stock ChromeOS w/RW_LEGACY
Fw Ver: Google_Dochi.14505.758.0 (04/19/2024)
Fw WP: disabled

MrChromebox, 2025-04-28:
added, but completely untested. Would only recommend flashing with a SuzyQ available to unbrick and a stock backup on hand

MrChromebox, 2025-04-28:
mind grabbing some debug logs if I post a debug build here?

logicone96, 2025-04-29:
Here is bios dump from your script (stock and modified).
Sizes of these dumps are different and stock is starting from empty data
https://drive.google.com/drive/folders/1VsymAbD0YorbPfk70Q4c5ygXVK8cJrsG?usp=sharing

MrChromebox, 2025-04-29:
the dumps aren't useful unfortunately, what I need is a serial log from a debug-enable build.

[coreboot_edk2-dochi-mrchromebox_20250429.zip](https://github.com/user-attachments/files/19960281/coreboot_edk2-dochi-mrchromebox_20250429.zip)

flash this debug-enabled build, and record the output of a single boot on /dev/ttyUSB1 using minicom or similar, then attach the full log here

logicone96, 2025-05-05:
Unfortunately, there is a problem, because I currently don't have a manual backup (dump), and it turned out that the script that did the backup messed up something and at the moment the laptop unfortunately doesn't work, and there is no bios anywhere on the Internet.

MrChromebox, 2025-05-05:
I don't see how that prevents you from flashing the above image and getting me the log I need to debug it. 

attach your stock backup (zipped) so I can see what's wrong with it

MrChromebox, 2025-05-05:
> this is the actual dump on which the hardware fails to boot after modification

again, this is useless to me. I need you to flash the debug ROM I posted above and collect the debug output on /dev/ttyUSB1 using a SuzyQ and minicom.

MrChromebox, 2025-05-05:
> [stock-firmware-DOCHI-20240419.zip](https://github.com/user-attachments/files/20038354/stock-firmware-DOCHI-20240419.zip)

there's nothing wrong with this backup. You flashed the `bios` region only as per the instructions, right?

logicone96, 2025-05-06:
> the backup you have is sufficient, when restored using the correct instructions.

How can I repair UEFI firmware with the backup made by your script? I can't do it via flashrom because I don't have a compatible programmer. I am using TL866II

MrChromebox, 2025-05-06:
oh, so you don't have a SuzyQ cable? I highly recommend getting one if you want to sort this out. I did say from the start that I only recommended flashing if you had one.

MrChromebox, 2025-05-06:
> is it now possible to repair this dump so that it can be programmed directly on SPI to fix the chromebook? Is it rather hard to do?

yes, using coreboot's ifdtool. You need to dump the IFD and ME regions from the UEFI firmware, and inject them into the stock firmware image, then write that image back to the board. You'll need to compile ifdtool and then it will be something like:

```
# dump all regions from UEFI firmware
./ifdtool -p adl -x <uefi firmware filename>`
#inject IFD and ME regions into stock firmware
./ifdtool -p adl -i fd:flashregion_0_flashdescriptor.bin <stock firmware filename>
./ifdtool -p adl -i me:flashregion_2_intel_me.bin <stock firmware filename>
```

then flash the stock firmware image back using your programmer

MrChromebox, 2025-05-07:
> I understand that Factory Shim/Provisioning Key etc. may be the problem,

why do you say that?

>  I understand that when I manually program the SPI bone with your firmware there will be no problem? Currently the laptop shuts down immediately after booting but from what I know the lack of any of the software components can be a problem.

without serial output from a SuzyQ cable, there is no way to debug the issue.

> Now it should work?

it should, assuming those are the correct ranges. you used ifdtool with the `-p adl` parameter, right?

logicone96, 2025-05-07:
> why do you say that?

I asked for a clean bios Acer Company and they wrote back that they can give it to me but it won't work without these tools - Factory Shim/Provisioning Key (clean factory bios).

> it should, assuming those are the correct ranges. you used ifdtool with the `-p adl` parameter, right?

Yes and from the uefi firmware file it pulled correctly “flashregion_0_flashdescriptor.bin”, “flashregion_1_bios.bin”, “flashregion_2_intel_me.bin” while it had a problem saving it in stock.rom (reported a problem that there is no such region). So I read where the FD, ME region is and copied it from the uefi firmware file and then uploaded it to stock.rom overwriting it without changing the file size. I'll check right away if this method works

MrChromebox, 2025-05-07:
> Okay now its working everything. I wanted this windows but now I prefer to leave it as is. To make windows I need this suzyq, yes? I will purchase this cable

since the UEFI image I provided you did not boot, yes you will need a SuzyQ to help me debug the issue. Or you can send me a device for testing =D