MrChromebox/scripts #474: script does not run on ubuntu

mikee512, 2025-05-03:
All the docs I read indicate that the script can run from chromeos and a linux install. I installed rw_legacy some time ago and wanted to upgrade to the latest firmware. I can't do this from chromeos because the sudo restrictions and VT-2 requires a password I do not recall setting. I dont want to powerwash and start over. When running the script from linux it initializes and presents the main menu. Selecting install/upgrade rw_legacy simply returns to the main menu. Looking at the script (firmware.sh) it seems isChromeOS is required to run flash_rwlegacy. Removing the condition for isChromeOS allows the script to install firmware. I don't get the latest. This is on an eldrid/volteer tgl hp x360. It did upgrade the fw to 20240729. Confused, as usual. Thanks for all the work you do! The reason I am looking to upgrade is to try and get the touchscreen working. I know this has been an issue for some time and I'll open an issue elsewhere for that.

MrChromebox, 2025-05-03:
> Selecting install/upgrade rw_legacy simply returns to the main menu. 

I can fix that

>  Looking at the script (firmware.sh) it seems isChromeOS is required to run flash_rwlegacy.

that's not correct, you're misreading it

> The reason I am looking to upgrade is to try and get the touchscreen working

RW_LEGACY is just a bootloader, it can't affect anything related to hardware. Your issue is with the stock firmware, not RW_LEGACY. Nothing I can do about hardware not working under RWL

mikee512, 2025-05-04:
The only other way to get to flash_rwlegacy is to use -U (unlockMenu(, right ? Yeah, I am looking at the ACPI differences between chromeos and the stock kernel. Yuck. Thanks for your help.

MrChromebox, 2025-05-04:
pushed a fix for the RWL flashing issue

mikee512, 2025-05-04:
fixed. thanks!

closed 2025-05-04 by mikee512