MrChromebox/scripts #475: Failed to download vilboz firmware

Bakurinha, 2025-05-05:
Hello, I'm trying to download the "vilboz" firmware, but without success. I even tried using the browser, but it returns several backwards "r". lol

MrChromebox, 2025-05-06:
what does this mean? how are you trying? the file exists on the server an is correct, so the script should not have a problem

Bakurinha, 2025-05-06:
I'm using a VM for the process, but I managed to solve it by using Windows itself and passing it to the VM.
I had tried to download the most recent firmware from the file, but both the direct link from the PC and "wget" didn't work, so I used the Windows "curl -o" command and it worked. Sorry for my bad English, I'm using a translator.

MrChromebox, 2025-05-06:
wget should have worked without issue, I've used it myself. must have been something odd with your VM networking setup

closed 2025-05-06 by MrChromebox