MrChromebox/scripts #478: I try to install the Full ROM Firmware, but firmware-util.sh is not working, and I tried to download the file ROM direct for website, but not work too. I tried to search alternative s but I failed. Is a coreboot_edk2-bobba-mrchromebox_20250430.rom

LinkIG7, 2025-05-22:


LinkIG7, 2025-05-22:
And I don't no if is still available but when I tried to access to mrchromebox.com/files for download the ROM I need, say that I don't have permission.

[image]

LinkIG7, 2025-05-22:
Well, I'm from México, and I bought a Chromebook in Marketplace for my University (because, you know, I'm poor 😅) so when take it for home, I see that is In enrollment enterprise. So I can't use it. But the problem I find is that the Chromebook is from United States Schools, so I can't return the Chromebook for US Schools and I have a not working Chromebook. So I thought to Install Linux but the developer mode is blocked and I can't do nothing (that's the reason that firmware-util is not work) so I see in YouTube a solution and I tried to use the CH143A programer to install Linux but I can't download the ROM that I need. I think that the page is not working so this is why I open the conversation. So thanks for the Reply.

MrChromebox, 2025-05-22:
> but I can't download the ROM that I need

what URL are you using to try and download the ROM?

LinkIG7, 2025-05-22:
The URL is "https://mrchromebox.tech/files/firmware"
That's the URL that the YouTube video give me to download the rom.

LinkIG7, 2025-05-22:
Oooh, thank you so much. The ROM works perfectly. So again, thankyyou for yourtime.

closed 2025-05-22 by LinkIG7