MrChromebox/scripts #489: HP Chromebook x360 14a Can I run chromebox via Fedora or Chrome OS (installed via RW Legacy)

MonsterSe7en, 2025-09-03:
Hey, 

Device : HP Chromebook x360 14a Intel version x86 64

Boardname : Bloog

Error Message : Chrome OS is missing or damaged. Please remove all connected devices and start recovery.

---

I enabled developer options and installed Fedora 42 Workstation on my Chromebook by RW Legacy. (Dualboot)

Touch and sound wasn't working on Fedora. I understand that you need to remove Write Protect and install a UEFI (Full ROM) Firmware in order to get that working.

## Concern

My Concern is that installing Fedora probably messed with the EFI or Chrome OS partition which leads to Chrome OS not working. 

## Question

So do I need to restore Chrome OS > Enable developer options and run the Mr. Chromebox script or can I just run the script via Fedora 42 (Installed via RW Legacy itself)

Thought I'd check before I remove the screw and unplug the battery (Remove Write protection)

Any Guidance is highly appreciated

archisman-panigrahi, 2025-09-03:
Press the spacebar. It will guide you on how to flash a usb drive with chromeos, for which you can use a windows computer).

After restoring chromeos detach the battery, run the script to install full uefi, re-attach the battery

MrChromebox, 2025-09-03:
You can flash the UEFI Full ROM firmware from Fedora, provided the script identifies the device properly.
If not, then you will need to perform a ChromeOS recovery, then flash the Full ROM firmware from ChromeOS.

MonsterSe7en, 2025-09-03:
I assumed as much. Decided to take the long tested route. I really really didn't want to brick a brand new chromebook.

And sir @MrChromebox YOU ARE A GENIUS!! JUST BRILLIANT!!!

I am unable to find the words to express my gratitude. 

The Chromebook is basically an unlocked laptop. Thank you!!

closed 2025-09-03 by MonsterSe7en